analogdevicesinc/m1k-fw #2: microoptimisation

itdaniher, 2015-03-30:
The M1K hardware is theoretically capable of supporting sampling rates well over the nominal 100KHz, with the max rate of the 7682 at 250K. Careful considerations of the main loops could allow operation up to the converter maximum.

phdussud, 2015-07-26:
The converter actually does 2 conversions per output sample (2 converters, 4 values per output sample). So at 100KS/s it does 200Kconversion/s, which is close to the max.

itdaniher, 2015-07-26:
That's correct that at two channels per sample, it's close to the '7682's max, but the wiring of the converters allows for them to operate interleaved on any particular analog signal, allowing a theoretical max (one signal) of 500Ksps.

phdussud, 2015-07-26:
Well, I believe you can't only measure 1 signal at max rate because the CONV signals of the 2 ADCs are tied together, so there is no way to cleanly interleave their conversion. You will only be able to achieve 250Ksps on 1 signal but you can measure 2 signals at 250Ksps as well.

itdaniher, 2015-07-27:
Ah, right. Thanks!

dbogdan, 2015-08-24:
The current version of the firmware supports 100 kSPS rate. This is the maximum rate that will be supported by the official software.

closed 2015-08-24 by dbogdan