analogdevicesinc/m1k-fw #3: "make" toasts the m1k!

pcercuei, 2015-04-16:
The Makefile will attempt to flash the m1k if it is connected via USB.

However, on my system (Debian testing), using the latest master branch, the flash will fail and result in a non-booting device, that makes a clicking noise, and don't appear on ```lsusb```.

This is fully reproductible and I already bricked two units.

The last few lines of my ```make``` log:
```
LN      m1000.elf
SIZE    m1000.elf
m1000.elf  :
section              size         addr
.text              0x354c      0x80000
.relocate           0x670   0x2007e000
.bss               0x1ac0   0x2007e670
.stack              0x800   0x20080130
.ARM.attributes      0x29          0x0
.comment             0x58          0x0
.debug_info       0x14837          0x0
.debug_abbrev      0x3a65          0x0
.debug_loc         0x7a15          0x0
.debug_aranges      0xe40          0x0
.debug_ranges      0x1c08          0x0
.debug_macro      0x1629c          0x0
.debug_line       0x10348          0x0
.debug_str        0x5a1a1          0x0
.debug_frame       0x1e60          0x0
Total             0xaabdb

   text	   data	    bss	    dec	    hex	filename
 0x3bbc	    0x0	 0x22c0	  24188	   5e7c	m1000.elf
OBJDUMP m1000.lss
NM      m1000.sym
OBJCOPY m1000.hex
OBJCOPY m1000.bin
/bin/sh: 1: [: -a: unexpected operator
Traceback (most recent call last):
  File "<string>", line 1, in <module>
AttributeError: 'NoneType' object has no attribute 'ctrl_transfer'
Makefile:2: recipe for target 'flash' failed
make: [flash] Error 1 (ignorée)
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/usr/lib/python2.7/dist-packages/usb/core.py", line 971, in ctrl_transfer
    self.__get_timeout(timeout))
  File "/usr/lib/python2.7/dist-packages/usb/backend/libusb1.py", line 819, in ctrl_transfer
    timeout))
  File "/usr/lib/python2.7/dist-packages/usb/backend/libusb1.py", line 552, in _check
    raise USBError(_strerror(ret), ret, _libusb_errno[ret])
usb.core.USBError: [Errno 5] Input/Output Error
Makefile:2: recipe for target 'flash' failed
make: [flash] Error 1 (ignorée)
python ./scripts/sam-ba.py
please wait...
Traceback (most recent call last):
  File "./scripts/sam-ba.py", line 26, in <module>
    dev.set_configuration(1)
AttributeError: 'NoneType' object has no attribute 'set_configuration'
Makefile:2: recipe for target 'flash' failed
make: [flash] Error 1 (ignorée)
```

itdaniher, 2015-04-16:
That bites! Good news is it's easy to unbrick a unit - short out the small jumper on the top of the board as per the m1k-fw README and use `bossac`. I'll look into this.

pcercuei, 2015-04-16:
Ok, I could fix them all.
Thanks!

lclausen-adi, 2015-04-16:
Should be 'if [-e ...]' instead of 'if [ -a ...'. Bash seems to ignore -a with just one expression, but dash throws an error.

Also which jumper exactly? There are 4 pads next to the PIO header.

itdaniher, 2015-04-16:
Short the smallest jumper. http://imgur.com/tOdcnSb

rgetz, 2015-04-16:
Can we add these instructions/pictures to the wiki? 

http://wiki.analog.com/university/tools/m1k/firmware

should exist, but currently does not.

-Robin

itdaniher, 2015-04-16:
@pcercuei I suspect this is related to https://github.com/analogdevicesinc/m1k-cad/issues/4. The script at master works perfectly on the test jig. I'll modify sam-ba.py to disable PB17, disabling all the power-hungry bits.

"Input/Output Error" isn't especially helpful, but sounds like the device is dropping off the USB bus for some reason. 

Short term, I'll drop the call to sam-ba.py in the 'all' Make target.

UweBonnes, 2015-04-20:
I am hit be a similar error. After calling "make" with pixelpulse running, the device was lost. After much searching and finally finding this thread and the picture  http://imgur.com/tOdcnSb, at least I can make the device enumerate as samba. But calling "python ./scripts/sam-ba.py" gives
please wait...
Traceback (most recent call last):
  File "./scripts/sam-ba.py", line 26, in <module>
    dev.set_configuration(1)
  File "/usr/lib/python2.7/site-packages/usb/core.py", line 559, in set_configuration
    self._ctx.managed_set_configuration(self, configuration)
  File "/usr/lib/python2.7/site-packages/usb/core.py", line 92, in managed_set_configuration
    self.backend.set_configuration(self.handle, cfg.bConfigurationValue)
  File "/usr/lib/python2.7/site-packages/usb/backend/libusb1.py", line 741, in set_configuration
    _check(self.lib.libusb_set_configuration(dev_handle.handle, config_value))
  File "/usr/lib/python2.7/site-packages/usb/backend/libusb1.py", line 571, in _check
    raise USBError(_str_error[ret], ret, _libusb_errno[ret])
usb.core.USBError: [Errno None] Unknown error

Only after the sequence: Short jumper, apply poser to the board, unplug and reapplu power, samba finally succeeded.

UweBonnes, 2015-04-20:
Sorry, didn't reread the last sentence. It should read:

Only after the sequence: Short jumper, apply power to the board, unplug and reapply power, samba finally succeeded.

itdaniher, 2015-04-20:
At 1dbc371, 'make' will no longer run 'make flash' by default.

I have not taken the time to debug this.

sam-ba.py works flawlessly with

> Linux debian-1638b 3.4.104-bananian #1 SMP PREEMPT Mon Apr 6 18:25:40 UTC 2015 armv7l GNU/Linux
> pyusb '1.0.0rc1'

UweBonnes, 2015-04-20:
Part of my problem was an "Powered USB Hub", where actual the power supply for the hub was broken and so the ADALM was only powered with 4 Volt. With that unpowered hub, also the negative rail didn't come up.

itdaniher, 2015-04-20:
@UweBonnes Youch, that would definitely cause problems. Can you run `uname -a` and let me know what kernel version you're running? Trying to gather information about what works / what doesn't work.

pcercuei, 2015-04-20:
On Debian testing, the ```python-usb``` package seems to be very old (0.4.3-1, probably using the old libusb API). By installing a more recent package from the "experimental" repository, flashing works fine.

itdaniher, 2015-04-20:
@pcercuei Glad to hear it.

UweBonnes, 2015-04-21:
>>>>> "Ian" == Ian Daniher <notifications@github.com> writes:

    Ian> @UweBonnes Youch, that would definitely cause problems. Can you run
    Ian> uname -a and let me know what kernel version you're running? Trying
    Ian> to gather information about what works / what doesn't work.

> uname -a
Linux hertz 4.0.0-rc6-1.g3062c5b-desktop #1 SMP PREEMPT Tue Mar 31 01:37:19 UTC 2015 (3062c5b) x86_64 x86_64 x86_64 GNU/Linux
-- 
Uwe Bonnes                bon@elektron.ikp.physik.tu-darmstadt.de

Institut fuer Kernphysik  Schlossgartenstrasse 9  64289 Darmstadt
--------- Tel. 06151 162516 -------- Fax. 06151 164321 ----------

itdaniher, 2015-04-23:
The makefile's been patched to not automatically try and dump FW. The sam-ba.py script has been confirmed to work by multiple individuals. It should be modified to check for pyusb version, but closing this issue.

closed 2015-04-23 by itdaniher