analogdevicesinc/m1k-fw #4: ChA Erroneous Data

itdaniher, 2015-04-20:
This has been mentioned elsewhere before - no fix known. Channel A (mostly current, occasionally voltage) shows regularly spaced erroneous samples, likely aligned to USB packet boundaries. This does not occur with stock (62.5k) firmware.

@dbogdan Would you take a look at data reliability across firmware compile toolchain? CSV export has landed in PP2 hopefully making such an analysis easier.

dbogdan, 2015-04-20:
@itdaniher, I'll take a look.

itdaniher, 2015-04-23:
Index and value of erroneous values: http://itdaniher.com/p?WDgG

itdaniher, 2015-04-23:
test script: http://itdaniher.com/p?onsV

itdaniher, 2015-04-23:
test script running against stock firmware: http://itdaniher.com/p?aFMm

itdaniher, 2015-04-23:
this issue is confirmed.

itdaniher, 2015-04-28:
@dbogdan Any updates on the issue? Do you need any additional test infrastructure? Once this is fixed, we'll set automatic firmware updates rolling.

dbogdan, 2015-04-28:
@itdaniher, it's on my to do list - tomorrow I will try to look at it.

dbogdan, 2015-04-29:
Using the original 100 ksps firmware (https://github.com/analogdevicesinc/m1k-fw/commit/f4d5abcc5ada965bf70ae1f1bd2e91f30d976165), I don't see issues with some older Pixelpulse2 builds (for example 2015-03-20). However, using the latest versions, the problem is present. Do you know what can cause this difference?

itdaniher, 2015-04-29:
What's the git hash of the build that appears to work? There has been nontrivial overhaul to both Pixelpulse2 and LibSMU over the last month, but given that the issue is reproducible with the provided test script, I believe the issue is in the firmware. 

Be wary of using Pixelpulse2 to identify the issue - the erroneous data seems confined to a single sample and is hard to spot from inside Pixelpulse2. CSV export functionality was not yet added in the 20/03/15 builds.

dbogdan, 2015-04-29:
I don't think that the problem can be reproduced with the test script if the first 100 ksps firmware is used.
It's almost sure that the latest firmware changes are also affecting the timing, but I wonder what else it makes things worse.
The working Pixelpulse2 has the 264cf0e hash.

itdaniher, 2015-04-29:
That version of pixelpulse2 does not support 100ksps. 

https://github.com/analogdevicesinc/pixelpulse2/tree/264cf0e

uses

https://github.com/analogdevicesinc/libsmu/tree/23a2a6e

https://raw.githubusercontent.com/analogdevicesinc/libsmu/23a2a6e/device_m1000.cpp

hardcodes the sample rate at the lower speed, not 100ksps.

Please test with the first 100ksps firmware, the test script provided, and a recent version of PySMU. Let me know if you need help setting up your test environment.

dbogdan, 2015-04-29:
But in the firmware the sampling rate is hardcoded to 100ksps (tc_write_rc(TC0, 2, 240);).
I'll try the test script.

itdaniher, 2015-04-29:
Ah, gotcha. That's not the case with newer firmwares - let me know how things pan out.

kevinmehall, 2015-04-29:
The garbage samples on chA voltage are proportional to chB current, and the garbage samples on chA current are proportional to chB voltage.

damercer, 2015-05-01:
When using my Python program the out of range samples happen on CH A current independent of anything and everything. They appear I'd say greater than 9 times out of 10 and always at the sample
points evenly spaced. And independent of the number of samples as longs as the length is greater than where the first bad spot is. Bad data happens in CH A voltage much less often but in the same places in the data record as the bad current data. This happened from day one with the first working 100KSP firmware.

itdaniher, 2015-05-06:
Unfortunately, this issue is present even on the purportedly "good" versions of Pixelpulse2. It's definitely a firmware timing issue.
[image]

phdussud, 2015-07-26:
I believe I know the cause. It is most likely due to a timing violation for the ADC. It says that all of the communication with the ADC needs to take place 1.2us (tData) after the CONV goes up. Well, the CONV goes up on TC0 RB and down on TC0  RC then the interrupt occurs (with latency), then the communication takes place. The RB is needlessly early (RC-10). I changed it to RC-2 and the problem went away. If someone else would confirm this, that would be great. 
					tc_write_ra(TC0, 2, 10);
					tc_write_rb(TC0, 2, udd_g_ctrlreq.req.wValue-2);
					tc_write_rc(TC0, 2, udd_g_ctrlreq.req.wValue);

itdaniher, 2015-07-26:
Thanks for the purposed fix - I'll duplicate and reply with results.

phdussud, 2015-07-26:
Thanks.

Patrick

https://github.com/analogdevicesinc/m1k-fw/issues/4#issuecomment-125034729

damercer, 2015-08-03:
If you guys are still looking to verify this fix, I have my FFT based
spectrum analyzer program which is very sensitive to even very
small errors at the LSB level ( 100dB dynamic range ) and would
be happy to test out a new binary if you have one to share.
With this program I'll know instantly if there are any errors in the data.

Either way I would love to start using a fixed binary.

Thanks.

Doug

rgetz, 2015-08-03:
Doug:

Dragos was working on this, but he is out on vacation last/this week - so he hasn't seen this thread, and is still out.

Hang on to early next week, and we should have an answer.

-Robin

itdaniher, 2015-08-03:
.@phdussud has sent me a patchset that offer functionality up to 100k, I'm still busy with family, but I will  push a branch with a working firmware .bin. -Ian

dbogdan, 2015-08-13:
The latest commit (https://github.com/analogdevicesinc/m1k-fw/commit/fa28b705e39d98068e5357d2bfb998108995bf1a) seems to help a lot - we will still need to run some additional tests to understand if the problem was completely solved or not.

phdussud, 2015-08-21:
Sorry for the delay, I was very busy at work. I can confirm that this firmware does not generate bad data during my tests. I would be interested in knowing why this is the case. It seems that the counter interrupt code is shorter (the test and return has been removed). Any other reason? 
 
Date: Thu, 13 Aug 2015 08:46:02 -0700

dbogdan, 2015-08-24:
Patrick, the software has tried to do too many things in very short time - sometimes, it couldn't finish all the tasks on time, so those bad samples appeared. Some operations were required just once, but they were executed for every sample - the last commit fixes that.

closed 2015-11-19 by dbogdan