analogdevicesinc/m1k-fw #5: Over current not handled properly

rgetz, 2016-02-23:
When an over current situation occurs in the M1k, things go crazy - they don't need to.

In the schematic - the ADM1177-1ARMZ (U2) handles over current of the device, and turns off power to the analog section.

The ADM1177-1 is connected to the ATSAM3 via I2C. I2C can periodically (~2 Hz) read the Status Byte to see if an overcurrent event has happened - and checking HS_ALERT (The hot swap operation has failed since the last time this was reset). If this has been set - the data transfer should stop, (you are sending garbage up the stack), and notify the stack that an over current has happened on the device.

This will effect libsmu as well - as it will need to handle this async event.

-Robin

dbogdan, 2016-04-05:
This is something that should be investigated: reading the ADM1177-1's status periodically might not be possible if the sampling rate is set to 100kSPS (too many things in very short time).

Dragos

rgetz, 2016-04-05:
In discontinuous mode - (repeated buffers) - we should be able to check at the end of the buffer. (maybe?)

-Robin

jonpearson, 2016-04-06:
While not a show-stopper, it would be good to get this resolved relatively quickly for the sake of the students that will be participating in the summer camp labs.

radhermit, 2017-02-10:
This is mainly working in recent fw versions and libsmu, it's just waiting on #11 now for some fixes.

radhermit, 2017-02-14:
Closing this in favor of #11 since overcurrent support now exists and works, it just requires a fix for full support.

closed 2017-02-14 by radhermit