analogdevicesinc/m1k-fw #7: the first two packets of data sent back to the host in SVMI or SIMV mode appear to be garbage

radhermit, 2016-11-29:
It seems like when a device is set by libsmu to be in SVMI or SIMV mode and then written to, the first two packets of data read back from the device don't match what is expected and instead contain garbage data perhaps because the data being sent to the device needs 2 packets worth of time to propagate through the system.

Is it possible on the firmware side to skip sending these two initial packets for those related modes?

damercer, 2016-11-30:
For non-continuous modes where the output samples and input  samples both start
and stop at the same point during repeated sweeps won't this misalign the two?

Also if this is the case why does it work with the current software (master libsmu)?

Sounds like there is a different problem here.... I would hesitate to change the firmware.

Doug

radhermit, 2016-11-30:
Yep, I figured out the issue in the library, closing this.

closed 2016-11-30 by radhermit