analogdevicesinc/m1k-fw #10: multi-device syncing has a nondeterministic delay until streaming starts

radhermit, 2017-01-30:
Running data captures across multiple devices after setting nonzero start of frames takes longer than expected. Each run takes up to a couple seconds and feels like a nondeterministic value instead of what should be done within a few hundredths of a second. This currently causes multiple device data captures to be practically unusable when allowing them to sync across a number of session captures due to the varied delay at the start of the capture.

itdaniher, 2017-01-30:
A few relevant lines at https://github.com/analogdevicesinc/m1k-fw/blob/master/src/main.c#L455

itdaniher, 2017-01-30:
When using multiple devices, a value "sufficiently far in the future to work on Windows" is selected for the SOF packet index, and the m1k waits until the present SOF is the specified SOF, then starts clocking out data. This gives userspace datastream alignment of within a sample, even on Windows....

radhermit, 2017-01-30:
How long does "sufficiently far in the future to work on Windows" mean? :smile:

It seems longer than necessary to me most of the time.

itdaniher, 2017-01-30:
https://github.com/analogdevicesinc/libsmu/blob/master/src/device_m1000.cpp#L439

The bitmask of 0x3c00 can be adjusted arbitrarily, if memory serves.

It's longer than necessary most of the time, and much longer than necessary on Linux. 

The current math yielded "acceptable" performance even on single-core atom CPUs, popular on aged netbooks, which are probably no longer a target platform for Pixelpulse2 / the M1K. @rgetz can probably advise further there.

itdaniher, 2017-01-30:
Also, as this codepath is only utilised for multiple devices, it was never optimized. I'm sure there's a better algorithm existent than "add a large fixed sized offset to the current SOF index and wrap," but it produced the synchronisation guarantee that I was working to engineer.

It's also worth noting that if two M1ks are plugged into different USB host controllers, this schema will not work as intended as SOF syncs in USB are only shared across the same bus. A single USB3 powered hub has enough bandwidth for more M1ks than one would reasonably use, and would be a solid recommendation to make for those looking to push the limits of what a single device can offer.

radhermit, 2017-01-30:
The main issue is the varied delay across numerous starts/stops of noncontinuous session runs done in quick succession, e.g. when running unit tests for multiple device captures. With the current implementation, syncing causes about 2 orders of magnitude slowdown or in specifics it takes about 0.4 seconds to run a multi-session script (examples/multi-cyclic-run.py) without syncing and ~40 seconds to run it with syncing enabled.

That amount of slowdown doesn't seem acceptable to me.

radhermit, 2017-01-30:
I should note I meant multi-run not multi-session and that those numbers were from a Linux host.

radhermit, 2017-02-13:
I should note, this does appear to affect continuous sessions as well on occasion causing the entire sample capture for certain devices to go missing (easily seen by running examples/plot-voltages.py multiple times). Disabling syncing (i.e. setting the SOF variable to 0) fixes this issue.

dbogdan, 2017-11-23:
The issue was fixed in libsmu: https://github.com/analogdevicesinc/libsmu/commit/41e2924b933b00e7732c18825d0ea98ec82cf04e

closed 2017-11-23 by dbogdan