analogdevicesinc/m1k-fw #8: add method of setting/storing the serial number for devices

radhermit, 2017-01-30:
We need a method of setting/storing/retrieving the serial number for a given device in order to properly support multiple device sessions on MATLAB.

itdaniher, 2017-01-30:
Device serial numbers (as currently implemented) are in ROM on the SAM3U.

dbogdan, 2017-01-31:
We already have the option of storing the calibration data in flash, so storing a custom serial too will be easy. However, the problem is that each reprogramming of the firmware overwrites the entire flash, so the calibration and the serial data will be gone.
Setting a different LED color for each connected board will help us identifying them. Isn't this a better option?

radhermit, 2017-01-31:
Reprogramming the firmware via the method done in libsmu doesn't reset the calibration. I'm somewhat sure we can make sure the serial data isn't wiped either.

dbogdan, 2017-02-13:
The option is now available:
https://github.com/analogdevicesinc/m1k-fw/commit/d6e6550fe75410f31ae235f39e03181251b91293
Example of using:
   dev.ctrl_transfer(0x40, 0x05, 0, 0, [1, 238, 2, 221, 48, 49, 50], 100)
The first 4 bytes are the SERIAL VALID key: 0x01, 0xee, 0x02, 0xdd and the remaining ones are the serial number (in this case 012????...).

radhermit, 2017-02-14:
Thanks, this is now supported in the libsmu rework branch and works well.

closed 2017-02-14 by radhermit