analogdevicesinc/m1k-fw #9: initialize the device in Hi-Z at startup

radhermit, 2017-01-30:
Currently, if you try grabbing samples on the either channel in HI-Z mode directly after plugging the device in sometimes the returned sample data starts much higher than normal (sometimes all the way at 5v) and then slowly works its way down to the normal value over a couple seconds depending on what value it starts at. If another set of data is captured directly after the first, sometimes it will still exhibit higher than normal values. Generally the values settle into normal ranges within a hundredth to a tenth of a second.

It doesn't occur on every device plug-in, but on enough to notice it's happening frequently. It also can cause test suite failures if you run the tests directly after plugging in a device, but running the tests a second time will pass.

It seems like it's possible to trigger the bad data on device plug-in more often when you touch the metal, micro-USB plug to something else electrically conductive before plugging it in. It can also be triggered by unplugging/re-plugging the host side of the USB cable, but that doesn't appear to trigger it as often. I've reproduced this on all supported OS's, on multiple host ports, with multiple USB cables, and on multiple devices.

The probable solution to fix the behavior above is to initialize the device into the Hi-Z mode during the right place at startup.

itdaniher, 2017-01-30:
IIRC, this is related to the settling time on the switchers. Likely best just to delay an extra 100ms on init.

rgetz, 2017-01-30:
@itdaniher : Do you remember which switcher?

As far as I can tell:
https://github.com/analogdevicesinc/m1k-cad/raw/master/02-037760-01-d-1.pdf

The "switchers" U9 (ADP1614ACPZ-1.3) and U5 (ADP2442ACPZ) are enabled on plugging the device into USB, and then are always up and stable. Soft start cap (0.01uF) on the ADP1614 should make it (1.23 V * 0.01uF / 5.5 μA) = 2ms - which is nothing. Maybe the issue is - this isn't big enough, and things are actually in current limit during startup. fixed soft start time on the ADP2442ACPZ is also 2ms, so I can't see this being the issue (should be stable by the time USB enumerates) @msuciu - can you check the supplies when things are plugging in? This should be an easy check - to find a unit that does this and then add a connection between U5.4 (PGOOD) and U1.57 (NRST of the SAM), and see if this fixes things or not.

@radhermit : when you say "directly after plugging" how quickly are you talking about?  within 1 second? or longer?

@msuciu : Can you try to replicate this - if it is a power supply startup issue can you look at replacing U6/U7 with LDOs that have power good outputs on them? (and then tying this node back to the SAM reset?)

Thanks

radhermit, 2017-01-30:
I've only tested up to about a minute and the condition still persists then, but I imagine it's until whenever the first query for samples is made.

rgetz, 2017-01-30:
If it is >1 second - its not likely a hardware issue, but it's something @msuciu can still report on.

-Robin

itdaniher, 2017-01-30:
@rgetz @radhermit I was mistaken, I mis-remembered the PWR_ENABLE drive configuration as taking place later in the init process, but as you correctly stated, it happens on device init at https://github.com/analogdevicesinc/m1k-fw/blob/b4d6d664e40cdef42076a632d9792d83fb07817a/src/main.c#L149. Comment withdrawn! 

Moving switcher-enable later in the process was a powersaving optimisation to reduce startup current until after host negotiation, but that's a part of the USB spec that I'm pretty sure has been obsoleted by smartphone charging.

itdaniher, 2017-01-30:
> Currently, if you try grabbing samples on the either channel in HI-Z mode directly after plugging the device in sometimes the returned sample data starts much higher than normal (sometimes all the [...]

I recall this being an exponential decay, which resonated with my memories about switcher enable. @radhermit can you confirm, maybe with a screenshot?

radhermit, 2017-01-30:
I'm not really working with pixelpulse but I can give you a simple dumped data plot using matplotlib.

[image]

radhermit, 2017-01-30:
With a more normal data capture plot looking like this without the decaying spur in the first few thousand samples.

[image]

msuciu, 2017-01-31:
Hi,

I try to reset the SAM with PGOOD pin from U5, than I increase the startup time for U6 and U7 and at the end I try to reset the SAM manually by connecting GND to NRST pin for a short period of time but the problem remain. I’m not shore if other LDOs with power good pin will solve the problem. I also leave a board for around half an hour without to start any capture and after that when I start the capture this bug appears. I still investigate this.

https://github.com/analogdevicesinc/m1k-cad/raw/master/02-037760-01-d-1.pdf

msuciu, 2017-01-31:
I try to reset the SAM with PGOOD pin from U5, than I increase the startup time for U6 and U7 and at the end I try to reset the SAM manually by connecting GND to NRST pin for a short period of time but the problem remain. I’m not shore if other LDOs with power good pin will solve the problem. I also leave a board for around half an hour without to start any capture and after that when I start the capture this bug appears. I still investigate this.

msuciu, 2017-02-08:
I tried to reset the SAM with PGOOD pin from U5, than I increase the startup time for U6 and U7 and at the end I try to reset the SAM manually by connecting GND to NRST pin for a short period of time but the problem remain. I’m not sure if other LDOs with power good pin will solve the problem. I also leave a board for around half an hour without to start any capture and after that when I start the capture this bug appears. After some investigations it seams that the ADC is not initialized before the first capture so if the ADC is initialized at startup this could solve the problem.

rgetz, 2017-02-08:
Can we try that in the firmware? I know Dragos is making some existing changes

dbogdan, 2017-02-09:
It seems that the issue is caused by the fact that the Internal Reference of the ADC is powered on only when the first capture is made. According to the ADC's specifications, the Turn-On Settling Time for the Internal Reference is typically 5 ms. We will try to power it on when the device is plugged in.

rgetz, 2017-02-09:
Dragos:

Awesome that you & Mihai were able to track it down to something in a datasheet. Now we can figure out an appropriate fix. 

- Robin

radhermit, 2017-02-10:
Just to note, I added a test for this in the python bindings at https://github.com/analogdevicesinc/libsmu/commit/5ab1af1dfc051fb7d09adadb239955195fb038b3 which assumes it'll be fixed for fw revs > 2.09. :smile:

It can be run manually via: ``./setup.py test -s -k test_read_initial_data`` in the bindings directory if you build the library and bindings and point LD_LIBRARY_PATH and PYTHONPATH to the correct locations as necessary.

dbogdan, 2017-02-20:
The issue was fixed: https://github.com/analogdevicesinc/m1k-fw/commit/a11c81f9224ae07970d8a7347dc65faad22c5782

closed 2017-02-20 by dbogdan