analogdevicesinc/m1k-fw #11: adm1177 overcurrent status needs to be reset for every data request

radhermit, 2017-02-07:
Currently the status isn't reset (via writing to bit 4 of the ALERT_EN register see http://www.analog.com/media/en/technical-documentation/data-sheets/ADM1177.pdf for details) so if an overcurrent event occurs several bits of the status byte used to determine when overcurrent events have occurred will be stuck on, regardless of whether an event occurred in the most recent sample acquisition or not.

This probably should be reset in config_hardware() which is run during device->on() in libsmu, i.e. at the start of every data capture request.

dbogdan, 2017-02-07:
I will take care of this.
Thanks,
Dragos

radhermit, 2017-02-07:
I should also note that the HS_OC bit does appear to map to the most recent overcurrent event, but being able to use the alert bit would be nice too.

radhermit, 2017-02-08:
Another question I have is why the write_adm1177() changes in https://github.com/analogdevicesinc/m1k-fw/commit/ed12ee17b3020976c33345489cabe031b0f1518b removed some of the options enabled in the command byte written.

Is there a way to get the ADC_ALERT and ADC_OC bits in the status byte to get properly toggled? I re-enabled the bits that were disabled in the command byte for write_adm1177() but that didn't change anything. I ask because it doesn't appear that the HS_OC bit is toggled in all overcurrent data acquisitions. This probably relates to the function definition for the HS_OC which defines differences between parts that latch off and parts that automatically retry.

I assume we're using the ADM1177 that is configured to do an automatic retry?

dbogdan, 2017-02-13:
We are only interested in finding out the status of the device - this is the reason why the option of reading the status was enabled and the voltage/current measurements were disabled.
In this version, HS_OC bit is set when an overcurrent condition appears and automatically cleared otherwise.
I agree that having a bit that can be cleared only manually would be a better option - however, I got the same results as you so I'll have to investigate this more.

radhermit, 2017-02-13:
Ah, makes sense, thanks.

dbogdan, 2017-02-14:
This is fixed now: https://github.com/analogdevicesinc/m1k-fw/commit/d1fc36e1f10069708c0d9b31a0851d29ee99f5d8
HS_OC bit acts the same: it is set when an overcurrent condition appears and automatically cleared otherwise.
Additionally, if HS_OC reports an issue, HS_ALERT is set too and cleared by config_hardware() (HS_ALERT reports if the hot swap operation has failed since the last time this was reset).

closed 2017-02-16 by dbogdan