analogdevicesinc/m1k-fw #17: Can the firmware be compiled (built?) using Windows computer?

damercer, 2018-12-11:
I have the following question. Can the firmware be compiled (built?) using a Windows computer? The instructions listed in the readme seem to be for Linux. Is that the only way (tool chain) to compile the firmware binary from source code?

If I can use Windows (7) what software do I need and where can I get it?

Thanks

Doug

commodo, 2018-12-11:
Building code on Windows is always tricky [compared to Linux], because you need to go find some toolchains to install.

I'd typically run a Linux VM to build something on Windows.

But, I guess you could also try other toolchains that are built for Windows; the situation seems far better now (i.e. there are windows gnu toolchains more readily available than 5+ years ago).
Some links:
* https://developer.arm.com/open-source/gnu-toolchain/gnu-rm/downloads
* I remember @mhennerich saying that he uses https://github.com/gnu-mcu-eclipse/windows-build-tools  
* http://gnutoolchains.com/

damercer, 2018-12-11:
Sorry if I'm asking very basic questions here but I'm starting below zero on the learning curve.

I have successfully installed the following:
From the first above link. gcc-arm-none-eabi-7-2018-q2-update-win32-sha2
From what little I was able to get by reading the readme that came with it, it seems like I also need something like Cygmin or MinGW for Windows. So I installed both:
mingw64-gcc4.7.1 and for cygmin setup-x86_64 Windows installer.
I downloaded a copy of the m1k-fw master files from here as well.
I'm not sure how to proceed from here.
Typing make in either the MinGW-w64 64-bit GCC build environment command window or Cygmin command line window does not seem to do anything. The PATH does seem to include the gcc-arm-none compiler but I'm at a loss as to what all is still missing.

Thanks

Doug

commodo, 2018-12-12:
[ I'm on a different timezone; so it takes some time to answer back ]

No worries.
I'm not sure if you need Cygwin or MinGW.

But, what you should do is go into the `m1k-fw` folder that you cloned from github.
You should see these files: `asf  config.mk  IO.mkd  LICENSE  m1000.bin  Makefile  README.md  scripts  src`

If you run `make` or `make flash` here, the `make` command should run and pick up the `Makefile` file.

Generally, running the `make` command implies there is a `Makefile` file in the location you are running the `make` command. Naturally, there are ways of overriding this default convention [of make picking up a `Makefile` file], but people rarely do, and we don't typically do it for our code repositories.

dbogdan, 2018-12-12:
Another Windows option would be to use Atmel Studio. I did that someday and it wasn't hard to include the source files into a new project. However, unfortunately, I didn't write anywhere the steps.

dbogdan, 2018-12-12:
By the way, the only toolchain that generated a usable 100 kSPS firmware was the one from CodeSourcery. No other options worked good enough.

damercer, 2018-12-12:
I was in the m1k-fw-master directory. The problem was that in both cases (Cygwin or MinGW) didn't recognize "make" as a valid command. Seems not everything to emulate a linux shell was there.

I'll take a look for CodeSourcery. Do you have a link?

Thanks

Doug

damercer, 2018-12-13:
I installed Atmel Studio 7.0.
As an experiment I ran it and created a new project as a test. When it asked for the processor board type I selected ATSAM3U1C which I think matches what is on m1k. I closed Atmel Studio and I copied the files from m1k master into the project directory. When I restarted Atmel Studio and opened the project the files seemed to be there and I could click and view them. But that is about as far as I got so far.

What do I need to do next?

Doug

damercer, 2018-12-13:
I've made some progress. I re-ran cygwin installer and selected all the packages to install and now have the make command. From cygwim terminal screen while in m1k-fw master (version 2.16) directory I ran make flash and it seemed to run without any major errors. It generated a m1000.bin file.

I flashed this new file to one of my m1k boards and it ran (read back as version 2.16)  but had the dropped samples problem in channel B. I compared the size of this new file with the original version 2.16 file. My new file was 15,992 bytes while the original is 16,384 bytes. So there is a difference.

How can I check to see if building from cygwin is using the correct version of the compiler? The one I just loaded as part of Atmel Studio 7.0?

Thanks
Doug

dbogdan, 2018-12-13:
Doug, I've added an Atmel Studio 7 project on the atmel_studio branch: 
https://github.com/analogdevicesinc/m1k-fw/blob/atmel_studio/m1k-fw.cproj
You can give it a try.
Most probably your generated file is correct - however, any smallest change (code, compiler, ...) will generate a different result - and most of the times something that doesn't meet the timing for 100 kSPS.

dbogdan, 2018-12-13:
By the way, on our boards we have the ATSAM3U2C microcontroller (not ATSAM3U1C).

damercer, 2018-12-13:
Yes. That Studio 7 project worked on my system. It generated a m1000.bin file of yet a different size. 16,064 bytes. It still had a few dropped samples at times on both channel A and B but not nearly as frequently as the version I built with Cygwin.

Very strange that this is so non-deterministic when creating the binary file. I don't know how anyone could use this in a commercial production environment.

Thanks

Doug

mhennerich, 2018-12-14:
The problem is that the main loop in terms of processing cycles is to tight on the edge, that a few assembly instructions, their placement, pipeline effects, optimization options, etc. distinguish between work or fails.

If we want to add extended functionally and this requires additions to the main loop, we can either decrease the sampling clocks or we optimize the main loop. As far as I can tell later has been already investigated. Dragos can give you some details on that.

damercer, 2018-12-14:
I don't believe that changing the initialization of the ADC Mux settings or the DAC channel addressing configuration, which can happen outside the main sampling loop before looping starts will change the timing. The file will be larger to add functions to select the configuration but the body of the sampling loop should not change. My intent is to experiment and see if just changing the ADC/DAC configuration bits even works.

By the way, I haven't mentioned this before because I have not fully investigated under what circumstance is happens, but samples are still sometimes dropped / lost even at lower sample rate settings even when using the current released fw version (2.16). It happens for certain sample rate settings and I've not looked close enough to say exactly what the conditions are. So from my perspective, base on my observations, simply lowering the sample rate (at least the way it is currently accomplished through libsmu) does not always guarantee proper functioning.

Thanks

Doug

dbogdan, 2018-12-14:
I'm posting here two screenshots made at the time I've been investigating the issue:

[image]

[image]

dbogdan, 2018-12-14:
I guess that we reached the microcontroller's speed limit. Even keeping the main loop the same but turning on an additional hardware peripheral (i.e., PWM) will break everything.

dbogdan, 2018-12-16:
My tests showed that the PWM can be always kept powered on (like it was when the RGB LED was used), but cannot be turned on/off during the runtime without breaking the timing. Because of this, I wasn't able to implement the feature requested on https://github.com/analogdevicesinc/m1k-fw/issues/14 .
The conclusion of everything I have tried until now is that implementing new features in firmware requires a sampling rate decrease.
The desired option is to still have only one firmware, but some features to be available only for certain rates. If this option can not be made functional, two different firmwares will be required.

rgetz, 2018-12-17:
Dragos:

Can we just make these functions available when capture is not active? I think that would be OK. 

- Robin

dbogdan, 2018-12-17:
OK, this is another option - we will mark it as the desired one.

damercer, 2018-12-17:
I have not been able to get even the present (master) version to compile without bad data appearing randomly in the data stream with the Atmel Studio 7 Windows tools. So I can't say if my changes have actually broken anything yet. But I have implemented options for the user to change the two ADC mux configurations and the DAC address bits on the fly. I have determined that it is indeed possible to run the ADCs with the mux sequencer off and get data for voltage only (or current) at 200 KSPS on each of A and B. And run data through just one DAC at a time at 200 KSPS. So configured as 2 scope channels and one AWG channel at 200KSPS.

In fact when the sequencer is off and the mux is fixed (configuration not changing between samples) to always read from one input there are NO bad samples in the data stream! My guess is that when the firmware is set to change the configuration back and forth between sequencer on ( 0x20F7) and off (0x20F1) the configuration write is randomly not finished before the next sample is taken (started?) thus the new setting is ignored and the sample is taken on the wrong input making that sample "bad" (as it says in the datasheet). 

So my current theory is that there needs to be a tiny bit more time between writing to the configuration register and the start of the reading of next sample. I've some ideas for more experiments to further confirm the root cause. I would not give up on getting this right yet.

Doug

damercer, 2019-02-10:
I'm closing this out because I've now gotten a version of the Atmel Studio 7 tools with the Mentor cross compiler working in Windows.

Thanks
Doug

closed 2019-02-10 by damercer