analogdevicesinc/m1k-fw #20: How to use Mentor Graphics Codebench to make firmware

damercer, 2019-01-13:
Hi Dragos:
I downloaded the Mentor Graphics Sourcery CodeBench Lite for ARM EABI for Windows from:
https://sourcery.mentor.com/public/gnu_toolchain/arm-none-eabi/arm-2014.05-28-arm-none-eabi.exe 
And installed it using the default settings at each step. It was added to the PATH variable.

My question is how do I use these new tools, with the Atmel Studio 7 project I've been using before? Do I need to set something different in Studio 7 or will it automatically change to these ones in Sourcery CodeBench?
I looked briefly in the how to guide that came up at the end of the install process but frankly that was mostly about how to start from scratch on an new project. There did not seem to be a GUI program in the folder where it was installed.

Any help would great.

Thanks
Doug

dbogdan, 2019-01-13:
Hi Doug,

All you have to do is to update the Makefile by setting the CC macro. Add something like this in the m1k-fw\Makefile:
   CC=C:\workspace\tools\Sourcery_CodeBench_Lite_for_ARM_EABI\bin\arm-none-eabi-gcc.exe

Make sure to update the path according to you installation.

Thanks,
Dragos

damercer, 2019-01-14:
Here is the changed make file:

build: all

flash: cleanLocal all
	@ if [ -e ./src/main.su ]; then echo -ne "\nMaximum Stack Size: $$(cat $$(find -name \*.su) | awk '{s+=$$2} END {print s}')\n\n"; fi
	@- python -c "import usb;usb.core.find(idVendor=0x0456,idProduct=0xcee2).ctrl_transfer(0x40|0x80, 0xBB, 0, 0, 1)" 2>&1 > /dev/null
	@- python -c "import usb;usb.core.find(idVendor=0x064b,idProduct=0x784c).ctrl_transfer(0x40|0x80, 0xBB, 0, 0, 1)" 2>&1 > /dev/null
	@ sleep 1
	- python ./scripts/sam-ba.py

cleanLocal:
	rm -f *bin *elf *hex *lss *map *sym *.o *.d *.su

CC=C:\Users\Doug\MentorGraphics\Sourcery_CodeBench_Lite_for_ARM_EABI\bin\arm-none-eabi-gcc.exe
HW_VERSION=0
GIT_VERSION=$(shell git describe --always --dirty='*')

include asf/sam/utils/make/Makefile.sam.in

cflags-gnu-y += -D'HW_VERSION=$(HW_VERSION)'
cflags-gnu-y += -D'FW_VERSION=2.30'

I did a binary byte by byte compare of two .bin files made before and after this addition to the make file and they only differed by one byte where I changed the version number from 2.20 to 2.30. I did a clean between building the firmware to force it to do it.

So either the two compilers are the same or I'm still not using the new one. I scanned over the messages it printed out while it built the second time and didn't see anything to indicate which compiler it was using.

Thanks

Doug

Doug

dbogdan, 2019-02-04:
Doug,

I was wrong, sorry. CROSS macro should be set instead of CC. So try:
CROSS=C:/Users/Doug/MentorGraphics/Sourcery_CodeBench_Lite_for_ARM_EABI/bin\arm-none-eabi-

Sorry again,
Dragos

damercer, 2019-02-04:
I changed the line in the Make file to: CROSS=C:\.... and now it fails with and error:

Severity	Code	Description	Project	File	Line
Error		recipe for target 'build/src/main.o' failed	m1k-fw	C:\Users\Doug\Documents\Atmel Studio\7.0\m1k-fw\asf\sam\utils\make\Makefile.sam.in	403

Not sure what this means?
Thanks
Doug

dbogdan, 2019-02-04:
Make sure you have "arm-none-eabi-" for CROSS instead of "arm-none-eabi-gcc.exe" as for CC. Did you keep the dash and remove the extension? Can you show me your entire line?

Thanks,
Dragos

damercer, 2019-02-04:
That fixed the build error. It made a new .bin file. I'll get back to you after I've had a chance to test this new binary on one of the m1k boards.
Thanks
Doug

dbogdan, 2019-02-04:
Perfect. Let me know the result.

Dragos

damercer, 2019-02-04:
The new m1000.bin file flashes to the board and runs. However when I run Alice I'm still getting some dropped / bad channel B samples (getting I data samples in the V data array) like the previous versions I build with the Windows tool chain. So there still must be some differences between these compilers.

Doug

damercer, 2019-02-10:
I was using a version that was not the final tested (on Linux tools). Using the same identical file in both the Linux tools and the Windows Atmel tools now produces identical binary files. Both compilers work perfectly on my changes to the firmware with no bad, dropped or missing samples,

It took same back and forth but it is all working now. Thanks.
Doug

closed 2019-02-10 by damercer