analogdevicesinc/m1k-fw #22: Problem building firmware on Ubuntu 18.04

damercer, 2019-02-02:
Hi Dragos:

I loaded VMware on this old Windows 7 64 bit ADI laptop I have and installed Ubuntu 18.04 on it. After installing the various prerequisites including what I think is the latest gcc-arm-none-eabi (using sudo apt install gcc-arm-none-eabi) I cloned m1k-fw master and did a sudo make flash. That seemed to work and produces a m1000.bin file (reads back as 2.16 as it should).
I then flashed that file to one of my Rev F boards. However when I run Alice I'm getting lots of dropped / bad samples (getting I data samples in the V data array) like the versions I build with the Windows tool chain.

What are you doing differently to get this to work properly?

Thanks

Doug

dbogdan, 2019-02-03:
Hi Doug,

On Linux, I'm using the same CodeSourcery Toolchain. Once you have it on your Linux machine, Makefile can be updated similar to https://github.com/analogdevicesinc/m1k-fw/issues/20 .

Dragos

damercer, 2019-02-04:
Hi:
I'm not a Linux user so I don't know much of the command line stuff. What is command used to "extract" the downloaded file? Where exactly should it be extracted to and how?
I think I set up a 64 bit version of Ubumtu 18.04 on VMware. That computer is not connected right now so I can't check any further details. Let me know of any other things that need to be loaded.
Thanks
Doug

dbogdan, 2019-02-04:
# Extract
analog@ubuntu:~/Downloads$ tar xvf ./arm-2014.05-28-arm-none-eabi-i686-pc-linux-gnu.tar.bz2
arm-2014.05/arm-none-eabi/
...
arm-2014.05/share/doc/arm-arm-none-eabi/html/libm/ccos.html

# Find the location
analog@ubuntu:~/Downloads$ pwd
/home/analog/Downloads

# export CROSS before running make
analog@ubuntu:~/Downloads/m1k-fw$ export CROSS=/home/analog/Downloads/arm-2014.05/bin/arm-none-eabi-

# or update the Makefile
analog@ubuntu:~/Downloads/m1k-fw$ sed -i '/GIT_VERSION/a CROSS=/home/analog/Downloads/arm-2014.05/bin/arm-none-eabi-' Makefile

Dragos

damercer, 2019-02-04:
I down loaded and extracted using tar inside /home/downloads.
looking in the file manager there seems to be a arm-2014.05 folder with things in it.
I changed the Make file in /home/m1k-fw like this:

[Makefile.zip](https://github.com/analogdevicesinc/m1k-fw/files/2829592/Makefile.zip)

Then from inside m1k-fw I do this and get an error:

doug@ubuntu:~/m1k-fw$ sudo make flash
rm -f *bin *elf *hex *lss *map *sym *.o *.d *.su
CC      build/src/main.o
make: /home/downloads/arm-2014.05/bin/arm-none-eabi-gcc: Command not found
make: *** [asf/sam/utils/make/Makefile.sam.in:406: build/src/main.o] Error 127

Any ideas?
Thanks
Doug

dbogdan, 2019-02-04:
Something is definitely wrong in your path. The user name is missing (you should have something like /home/username/Downloads) and the name of the downloads folder most likely starts with an uppercase character (Downloads). Linux is a case sensitive operating system.

Dragos

damercer, 2019-02-05:
OK, in a terminal screen I navigated to where the files were extracted and did a pwd:
/home/doug/Downloads/arm-2014.05/bin
I then cut and pasted this text letter for letter into the Make file adding the arm-none-eabi- to the end:
CROSS=/home/doug/Downloads/arm-2014.05/bin/arm-none-eabi-
And I'm still getting the same error message when I try make in the m1k-fw folder?
I'm not the best typist when using the terminal command line but I think I'm doing this right. (a major reason I don't like using Linux)
Still not the answer to the problem I think.
Thanks
Doug

dbogdan, 2019-02-05:
The new path seems correct. Can you install lib32z1, lib32ncurses5 and try to run make again (sudo apt-get install lib32z1 lib32ncurses5)?
Thanks,
Dragos

damercer, 2019-02-06:
Adding those libraries fixed building the firmware binary. Thanks.
I also flashed this new binary to an m1k board and this one now seems to function as well as the one in master (2.16). I'm not seeing any bad, skipped, dropped, missing samples in the data using ALICE.
So I have this version working like yours in Linux.
Would adding these libraries (if possible to do) to the Atmel Studio 7 tool on Windows fix that one as well? It would be great if that could be made to work as well.
Thanks again.
Doug

dbogdan, 2019-02-06:
No, those libraries were required only because you are on a 64-bits machine where you want to run a 32-bits application (the Toolchain).
It's a little bit strange that you get a different result on Windows. Do you use the same 2014.05-28 version?
Thanks,
Dragos

damercer, 2019-02-10:
I guess this is all resolved now. Please see my pull request #21 for the changes I'd like added to the firmware.
Thanks.
Doug

closed 2019-02-10 by damercer

F1r3Hydr4nt, 2025-03-22:
> Download and extract: https://sourcery.mentor.com/public/gnu_toolchain/arm-none-eabi/arm-2014.05-28-arm-none-eabi-i686-pc-linux-gnu.tar.bz2

> Dragos

Have you got that file? Could you share it if so, or anybody else for that matter?