analogdevicesinc/m1k-fw #24: Output voltage toggling on and off in source voltage mode

thedude8, 2019-10-11:
When doing a constant voltage in SVMI mode, I am seeing the voltage toggle from the command voltage output value and zero every 100ms or so. I am using the latest firmware (2.16 & 2.17) and latest PixelPulse2 software. I tried this using Alice as well with the same results. I also tried this out on multiple ADLM1000 devices and computers, with the same behavior showing on both.

I also observed that moving the PixelPulse2 software from background and foreground changed the duration time from 100ms to maybe 500ms or so, if that helps debug this.

Anyone else experience this before?

cristi-iacob, 2019-10-11:
Hi!

Sorry, but we cannot reproduce this bug. Can you please give us more information about this issue? 

1. What operating system(s) are you using? 
2. What are your PC's specifications?
3. What settings do you have in Pixelpulse when this bug appears? ((non-)continuous mode, sample time)
4. Can you please attach a screenshot with the exact bug's behaviour? 

Thanks!

sirpatil, 2019-10-11:
Thank you for quick response. We figured it out, by default "repeated sweep" mode is enabled in  PixelPulse2, after turning it off it worked as expected.

closed 2019-10-14 by cristi-iacob