analogdevicesinc/m1k-fw #26: Current version number?

peterbuxton, 2020-01-22:
PixelPulse 2 running on Windows 2 reports that my ADALM1000 firmware is ver 2.16 and up to date. Alice 1.3.4 in Linux with python 2.7.17 reports a warning: This ALICE version Requires Firmware version > 2.16. When I click OK, it asks to flash the device. I answer Yes, and it finds m1000-2.17.bin in the alice-1.3.4 directory. 
1. Does that mean that PixelPulse 2 is not up to date? 
2. What is the version on this GitHub page? I don't see a current version number.
3. Is it safe to flash the device with 2.17 using Alice? (using PixelPulse is recommended in ADI/TOC)

peterbuxton, 2020-01-23:
https://github.com/analogdevicesinc/Pixelpulse2/releases/tag/v1.0.4 is the source from where I downloaded and installed PixelPulse on Windows 10. The About screen in PixelPulse shows:
--------
Built: 2018-09-05 Version: v1.0.3-3-gd3c6663
The most recent release is 1.04, published at Wed Sep 5 07:21:36 2018 GMT-0700.
It is available for download at https://github.com/analogdevicesinc/Pixelpulse2/tag/v1.0.4.
--------
As I said in the first post, the Device Manager showed ADALM1000 firmware ver 2.16 and up-to-date. I went ahead and let Alice flash it to ver 2.17F and now PixelPulse correctly shows ver 2.17F and says it is up-to-date. So my current issues are:
1. PixelPulse About shows v1.0.3 and suggests upgrade to v1.0.4 but that is already what I installed.
2. PixelPulse said before that firmware ver 2.16 was up to date, and also says it now for ver 2.17. It looks like up-to-date can be either (any?) version.
3. Should the version number be visible here on GitHub? Maybe it doesn't matter if the software automatically upgrades it, but Alice used a later one than PixelPulse.

peterbuxton, 2020-01-27:
After some testing, the only firmware issue out of the above comments is just that this GitHub page doesn't show what firmware version is available to download, if that matters. Alice and PixelPulse2 both flash whatever file they have installed with them properly to the M1K. It just seems there should be something written about the versions here, since I've encountered 2.16 and 2.17 so far. I will also post a comment on the PixelPulse page with an update about version conflicts I mentioned here.

AlexandraTrifan, 2020-01-27:
Hi,

Pixelpulse2 checks the m1k-fw page automatically for the latest release ( https://github.com/analogdevicesinc/m1k-fw/releases ) if your PC is connected to the internet. The firmware is not bundled in the PixelPulse application, but seems to be bundled in Alice.

We created the latest official release, v2.17 (using the same source code as **Alice's m1000-2.17.bin**), so PixelPulse2 should be able to find it and update the board automatically. Alternatively, it can be downloaded from the page mentioned earlier.

Thanks,
-Alexandra

peterbuxton, 2020-01-27:
Thank you for the correct link address. That solved the problem.

The **ADALM1000 Overview** page
(https://wiki.analog.com/university/tools/m1k) section **Device Software/Firmware** has a link titled **M1K Microcontroller Firmware Source** links (incorrectly?) to one level up from /releases, which does not show the version. Therefore, that's why I couldn't find the version number. 

-Peter

closed 2020-01-27 by peterbuxton